breakintoprogram/agon-bbc-basic #2: The sprites_1.bbc demo sometimes fails to properly clear the screen. Garbage stays behind from previous things that were on the screen earlier (I sent you a picture via Twitter DM), and the garbage seems to be treated like sprites too, as it is not overwritten.

TheByteAttic, 2022-10-02:


breakintoprogram, 2022-10-03:
*FX 19 is currently implemented using a HALT; this could be being tripped by other interrupts on the system. Modify to change change in TIME counter.

breakintoprogram, 2022-10-04:
This was to do with the order I was setting up the screen resolution in set_mode. Amended to:

1. Change the video resolution
2. Assign and initialise a new canvas to the video controller
3. Clear the canvas (belts-and-braces)
4. If there are any sprites running from last session then remove them (not set # to 0)
5. Do a sprite refresh to clear them off screen (more belts-and-braces)

Also in the VDU sprite code, added a condition to call removeSprites instead of setSprites if # sprites is 0

closed 2023-03-20 by breakintoprogram