breakintoprogram/agon-bbc-basic #5: In the sprites_1.bbc test, one or two sprites always seem to flicker in an unstable way, unlike the rest.

TheByteAttic, 2022-10-02:


breakintoprogram, 2022-10-03:
This may be down to the sprite catching the raster, or an errant interrupt. BBC BASIC uses the *FX 19 command to wait for the vertical blank interrupt. It is currently implemented using a HALT instruction which is a bit lazy as it may also proceed on other interrupts, i.e. UART. Instead check the LSB of _clock as this ticks on every vertical blank interrupt.

breakintoprogram, 2022-10-03:
Implemented the fix in BBC BASIC, still seems to get some sprite tearing. Will investigate further.

breakintoprogram, 2022-10-03:
Possible solution is to write the sprite move code in assembler to increase max number of sprites that can be moved per frame.

breakintoprogram, 2022-10-04:
Fix in OSBYTE_13 (*FX 19) function to wait for clock increment instead of doing a HALT locks up Bernado's units.

breakintoprogram, 2022-10-17:
There was a bug in my previous fix for OSBYTE_13 which has now been fixed. The flickering is just a function of the FabGL libraries and a limitation of the ESP32, so marking as 'Won't Do'.

closed 2023-02-20 by breakintoprogram