breakintoprogram/agon-bbc-basic #31: MOS commands that try to access file freeze when launched from Basic

lennart-benschop, 2023-03-26:
When I try to run a MOS command from Basic, as soon as it attempts to access a file, it freezes. Ctrl-Alt-Del won't work either.

Commands like this no longer work when launched from Basic.

*more file
*comp file1 file

The problem started with the code checked in during 2023-03-26, code checked out at the start of this day, worked fine.
They still work fine when I launch them directly from the MOS prompt. The programs I try to run are from my agon-utilities repo. They are 24-bit mode programs (ADL=1) and I take care to set MB to 0 and restore the original value when I exit the program. This has always worked correctly since I developed these and I could always run them from Basic.

breakintoprogram, 2023-03-26:
Not entirely limited to file commands:

- memdump: crashes
- disassemble: works fine
- hello_16, hello_24, hello_c: work fine

breakintoprogram, 2023-03-26:
Root cause analysis:

The interrupt in BBC BASIC calls a function in the BBC BASIC segment (&4xxxx). This assumes that MB will be fixed at 4, which is not the case if a MOS function is loaded in to run. A temporary solution has been implemented where MB is saved, set to 4 before calling that code, and then restored.

closed 2023-04-02 by breakintoprogram