breakintoprogram/agon-bbc-basic #44: Enhancement: Provide ASCII keycode support in GET($)/INKEY($) commands to detect keypresses of Fn keys and cursor keys

tonedef71, 2023-05-03:
Currently, Agon Basic commands like GET and INKEY return a zero for key presses such as the F1 through F12 keys and the individual cursor keys.  It would be useful if ASCII code values were returned when pressing these special keys.  

I'm not sure what the ASCII code mappings should be for these keys with non-printable characters; there does not appear to be any particular standard to follow.  Some possible candidates: 

[1] Re-use any FabGL ASCII code mappings for the extra keys
[2] Emulate the ASCII code mappings from a popular BBC Basic system (e.g. the Acorn BBC Micro)
[3] A brand new ASCII code mapping unique to the AgonLight

leventp, 2023-05-03:
Just tested. BBC Micro does not return any values for the function keys. They are used to define shortcuts to keywords. I already created a ticket to enable *KEY  functionality.

tonedef71, 2023-05-03:
>> Just tested. 

That's cool that you have access to an actual BBC Micro to compare and test with.

>> BBC Micro does not return any values for the function keys. 

Agon BASIC does appear to return a value for each Fn key press; that value just happens to be zero.  In the case of INKEY, when no key is pressed within the specified time frame, the value 255 is being returned.

>> I already created a ticket to enable *KEY functionality.

Excellent.

I still maintain that it would handy (and completely in line with other implementations of BASIC) that a unique ASCII code be returned for each Fn key press.  Even when the Fn keys have been redefined to return a keyword shortcut, you still might want to know which Fn key got pressed while you are polling for individual keypresses in a BASIC program with GET/INKEY.

leventp, 2023-05-03:
BBC Basic for Windows defines the key values for F1-F12 from 145 to 156.  This  range could be used as a reference.
Hacked the  do_keyboard function in video.ino file  (agon-vdp)  with the following and worked OK. Sure Dean knows the best way of adding this functionality.
...
	case fabgl::VK_BACKSPACE:
	  keycode = 0x7F;
	  break;
        case fabgl::VK_F1:
          keycode = 145;
          break;
        case fabgl::VK_F2:
          keycode = 146;
          break;
	      case fabgl::VK_F3:
          keycode = 147;
          break;
        case fabgl::VK_F4:
          keycode = 148;
          break;
        case fabgl::VK_F5:
          keycode = 149;
          break;
        case fabgl::VK_F6:
          keycode = 150;
          break;
        case fabgl::VK_F7:
          keycode = 151;
          break;
        case fabgl::VK_F8:
          keycode = 152;
          break;	
        case fabgl::VK_F9:
          keycode = 153;
          break;
        case fabgl::VK_F10:
          keycode = 154;
          break;
        case fabgl::VK_F11:
          keycode = 155;
          break;
        case fabgl::VK_F12:
          keycode = 156;
          break;		
	default:
	  keycode = item.ASCII;	
	  break;
...

astralaster, 2023-05-03:
This code shows how you can already achieve the behaviour with the current version of bbc basic. I think we should change GET and INKEY to something like this:
`   10 PRINT "Last key:",FN_getKey(1000)
   50 GOTO 10
   80 :
   90 DEF FN_getKey(timeout%)
   91 A%=&A0 : L%=&19 : keycount%=USR(&FFF4)
   92 syncs%=timeout%/17
   93 R%=-1
   94 FOR I=1 TO syncs%
   96   L%=&19
   97   IF keycount%<>USR(&FFF4) THEN L%=&05 : R%=USR(&FFF4) : IF R%=0 THEN L%=&17 : R%=USR(&FFF4) ELSE *FX 19
  101 NEXT
  120 =R%
`

tonedef71, 2023-05-03:
> Changing the do_keyboard in ESP has the disadvantage, that it is rendered back on the VDP. So when you press function keys you will see empty chars or weird symbols.

Is there somewhere within the VDP code which can be modified to not display anything to the display when the keycode is a function key?   That should handle the scenario you describe.

And if the pressed function key has been redefined to be a macro of commands to be displayed (a requested enhancement), then there would be other code in the VDP to handle that particular scenario.
 
>I think we should change GET and INKEY to something like this:
```
10 PRINT "Last key:",FN_getKey(1000)
50 GOTO 10
80 :
90 DEF FN_getKey(timeout%)
91 A%=&A0 : L%=&19 : keycount%=USR(&FFF4)
92 syncs%=timeout%/17
93 R%=-1
94 FOR I=1 TO syncs%
96 L%=&19
97 IF keycount%<>USR(&FFF4) THEN L%=&05 : R%=USR(&FFF4) : IF R%=0 THEN L%=&17 : R%=USR(&FFF4) ELSE *FX 19
101 NEXT
120 =R%
```
I'm curious how line 92 was arrived at... 17 is an intriguing divisor.

Is the timing loop of this sample code as sub-second accurate as the timing of the `INKEY` command?

astralaster, 2023-05-04:
That's the frame timing in Ms. So timeout is expected to be Ms. *FX 19 is waiting for the vsync and this should be 16.7 Ms. Maybe it's better to use timers here.

tonedef71, 2023-05-04:
> That's the frame timing in Ms. So timeout is expected to be Ms. *FX 19 is waiting for the vsync and this should be 16.7 Ms. Maybe it's better to use timers here, because one mode is not 60hz.

Ok.  So the divisor 17 comes from 1000 ms divided by 60 frames per second (16.66...7) rounded up.  In my build of the VDP, I am using a modeline setting for 300x200 which refreshes at **_70 Hz_**.  If the system clock would be more precise and more normalizing, then that would probably be the way to go.

leventp, 2023-05-04:
I still think it is better to assign a  range of code for the function keys. There is nothing wrong If  pressing a function key displays the corresponding character code.  BBC Basic for Windows does it this way (and of course provides setting them with *KEY)

astralaster, 2023-05-04:
> > That's the frame timing in Ms. So timeout is expected to be Ms. *FX 19 is waiting for the vsync and this should be 16.7 Ms. Maybe it's better to use timers here, because one mode is not 60hz.
> 
> Ok. So the divisor 17 comes from 1000 ms divided by 60 frames per second rounded up (i.e. 16.66...7 --> 17). In my build of the VDP, I am using a modeline setting for 300x200 which refreshes at **_70 Hz_**. If the system clock would be more precise and more normalizing, then that would probably be the way to go.

I've looked it up and the timer is tied to the vsync as well:
sysvar_time:		EQU	00h	; 4: Clock timer in centiseconds (incremented by 2 every VBLANK)

tonedef71, 2023-05-04:
> I've looked it up and the timer is tied to the vsync as well: sysvar_time: EQU 00h ; 4: Clock timer in centiseconds (incremented by 2 every VBLANK)

So that would be an internal refresh rate of 50 frames per second?

breakintoprogram, 2023-11-15:
The function keys, along with other non-ASCII keys, can now be checked using INKEY(-n) in BASIC.

https://github.com/breakintoprogram/agon-docs/wiki/MOS-API-%E2%80%90-Virtual-Keyboard

closed 2023-11-15 by breakintoprogram