breakintoprogram/agon-bbc-basic #46: Support some frequently used OSBYTE calls using USR &FFF4

lennart-benschop, 2023-05-09:
Some BASIC programs (especially those using MODE 7 for teletext graphics) use an expression like 

(USR &FFF4 AND &FF00) DIV 256

to read the character on the screen under the cursor. This is equivalent to 

ASC(GET$(POS, VPOS))

Could this (and maybe a few other frequently used OSBYTE calls) be implemented? BBC Basic for DOS and BBC Basic for SDL do implement this functionality.

breakintoprogram, 2023-05-19:
@lennart-benschop is this OSBYTE &87?

https://beebwiki.mdfs.net/OSBYTE_%2687

lennart-benschop, 2023-05-19:
Indeed this is OSBYTE &87, as is used in the example at the end of Chapter 28 of the original BBC Micro User Guide to read chars from the screen. Also used in MODE7DEM included with BBC Basic for SDL.

breakintoprogram, 2023-05-19:
Sample code
```
10 A%=&87
20 X%=USR(&FFF4)
30 REM L: Character under cursor
40 REM H: Current screen mode
```

lennart-benschop, 2023-05-21:
Returning the current screen mode by the VDP is certainly useful for programs that want to return to the screen mode they were called from.  For example a game that wants to run in MODE 2 and when finished returns to whatever mode it was called from.

closed 2023-06-27 by breakintoprogram