breakintoprogram/agon-bbc-basic #47: BASIC programs saved as BAS which contain empty lines do not load properly into Agon BASIC.

tonedef71, 2023-05-09:
BASIC programs saved as BAS which contain empty lines do not load properly into Agon BASIC.

A saved BASIC program listing which resembles the following ...
```
     5 :::::::::::::::
    10 : REM Hello W :
    15 :::::::::::::::
    20
    25 PRINT "Hello there, World!"
    30 GOTO 25
```
... would load into Agon BASIC as the following:
```
     5 :::::::::::::::
    10 : REM Hello W :
    15 :::::::::::::::
```

This gives the impression that when the `LOAD`er encounters a blank line, such as in `Line 20` above, the `LOAD`er believes that the EOF has been reached.

Currently, such BASIC program listings could only be authored outside of Agon BASIC (such as in Nano or in the BBC BASIC IDE), since it is not possible within BBC BASIC to enter a line number without any keywords or tokens and press `Enter` without it being interpreted as manually deleting a line of BASIC code (whether it exists or not in the listing.

Interestingly, a BBC tokenized program with an empty line does LOAD into Agon BASIC without issue, which leads me to believe that support for BAS program listings with empty lines is merely a limitation of the text loading support of the `LOAD` command.  

`SIDE NOTE:` If folks wish to be able to enter empty lines of code within Agon BASIC, then the BASIC/MOS editor would need to be enhanced to support the entry of empty line numbers, but it should only support that behavior when the line number does not already exist, otherwise support for manually deleting a line of BASIC code by entering its line number would be broken.  An alternative way to support this behavior within Agon BASIC might be to enforce a definition that an _empty line of BASIC_ is a line number followed by _at least one space_.

oldpatientsea, 2023-05-10:
Note, you can use a statement separator ```:``` on its own to space out listings, eg:

```
     5 :::::::::::::::
    10 : REM Hello W :
    15 :::::::::::::::
    20 :
    25 PRINT "Hello there, World!"
    30 GOTO 25
```
The examples in the BBC BASIC (Z80) manual use this technique for empty lines but it doesn't explicitly recommend it as a convention for making a more readable listing. I can add a note to that effect.

breakintoprogram, 2023-05-19:
Fixed bug in ONEDIT1 - it would reset to command line (CLOOP) rather than returning if a line with only a line number was entered.

closed 2023-06-27 by breakintoprogram