breakintoprogram/agon-bbc-basic #49: sprites_4.bas causes VDP to shut down after several minutes

jonathan-teh, 2023-05-30:
I suspect this is a VDP bug but since the reproducer is in BBC BASIC I've raised it here. Straightforward to reproduce:
```
LOAD "SPRITES_4.BAS"
RUN
```
For me, after between 1 min 30 s and 5 minutes (across 4 runs), VDP appears to shut down, the monitor no longer detects a signal and shuts off, the keyboard is unresponsive to any input e.g. to toggle caps lock or num lock. Pressing the reset button is the only way back to a usable machine.
`sprites_5.bas` can also cause a shutdown but takes longer, 12 minutes on the first run, the second run lasted 16 minutes before I stopped it.
MOS 1.03, VDP 1.03, BBC BASIC 1.04 on an Agon Light 2. Has been reproduced on another Agon Light 2 running the same software versions.

AntonioCarlini, 2023-05-31:
I can reproduce this easily with no special effort. Boot into BBC BASIC, "LOAD TESTS\SPRITES_4.BAS", "RUN" and wait.
Reproduced fives times out of five attempts:
  Trial A: Unresponsive by 10 mins
  Trial B: Screen blanked after 3mins
  Trial C: Screen blanked @ 1m09s
  Trial D: Screen blanked @ 10s
  Trial E: Still running @ 9m30s, but screen blanked by 30m

In Trial A I set it of and only came back after 10m; in trial E I gave up watching after ~10m but when I came back after 30m the screen was blank. For B, C and D I sat and watched with a timer running on my phone.

AL2 with MOS 1.03, VDP 1.03, BBC BASIC 1.04

lennart-benschop, 2023-05-31:
I get the same crash on the original Agon Light (one that was assembled by Bernardo himself), with sprites_4.bas, and also with sprites_5.bas

jonathan-teh, 2023-06-01:
Trying to gather more data points, I changed line 960 to `MODE 3`  (640x480) and it survived 30 minutes until I manually stopped it. Thinking it might have something to do with sprites moving offscreen, I changed these lines:
```basic
   40 SW%=320-16
   50 SH%=200-16
```
but with the usual `MODE 2` it crashed after 2 minutes.
I then tried `MODE 1` (512x384) and it crashed after 20 minutes.

jonathan-teh, 2023-06-04:
Looking around, a similar bug was fixed for VDP 1.03 RC3 in breakintoprogram/agon-vdp#28 by breakintoprogram/agon-vdp@94220dee285dc08cd6850d0e59edf3d4b5689ac9 (perhaps a link keyword would make it easier to find) but prior to that it was easier to reproduce with `sprites_5.bas` rather than `sprites_4.bas`.
Then perhaps it regressed in VDP 1.03 RC5 with breakintoprogram/agon-vdp#40 from breakintoprogram/agon-vdp@1da15096e39122dc242322b78350ebcef793c396 and breakintoprogram/agon-vdp@ba597c8fb9b5e40d8188ddd0e7be91fe50e27313.