breakintoprogram/agon-bbc-basic #54: Consider Displaying BBC BASIC (Agon) Version info on startup

oldpatientsea, 2023-06-16:
[image]

BASIC currently shows the version number for the BBC BASIC (Z80) code on which it has been based rather than the BBC BASIC (Agon) version number which is more important for determining if a user has the correct version of BASIC for their firmware.

If it showed both version numbers on startup, this might help speed up the process of finding answers when new users post problems on github, facebook, discord, etc

ZbigniewB, 2023-09-05:
They are actually version (3.00) and **release** (1.05) numbers.

envenomator, 2023-10-15:
I've seen too many people run into problems, thinking they were on the latest release of bbcbasic for the Agon, while actually they were running an older one. It would really help users to display something like
'BBC Basic (Z80) Version 3.00, Agon release 1.0x' at each startup

closed 2023-11-15 by breakintoprogram