breakintoprogram/agon-bbc-basic #61: *BYE not always works

ZbigniewB, 2023-08-31:
Usually it works, but try to do the following:
— load "triangles.bas" from "tests" sub-dir
— run it, then press Esc to break
— after I typed *BYE it's not going to let me back into MOS; it seems to be hung

stevesims, 2023-09-01:
It may be worth updating to version 1.05 of BBC BASIC, as well as the latest release candidates for MOS and the VDP.

I can't reproduce the error you're seeing on my machines, which are running later MOS, VDP and BASIC.

breakintoprogram, 2023-09-04:
Hi @ZbigniewB please let me know if using the latest and greatest fixes the issues you are having.

- MOS 1.03 / VDP 1.03
- BBC BASIC 1.05

As far as I am aware, this bug was fixed in #33

ZbigniewB, 2023-09-04:
Yes, I'm using the versions you mentioned — a while ago I downloaded again BASIC 1.05 to be 200% sure it's that release (maybe it would good idea to complete its startup message to e.g. „BASIC V3.0 R1.05”, or the like?) — and did that test again. No, in the situation described it doesn't let me back into MOS prompt. The cursor blinks, but there's no „star” prompt.

breakintoprogram, 2023-11-15:
Okay, back onto this now. How are you launching BBC BASIC.
It should be (from MOS)
```
LOAD bbcbasic.bin
RUN
```
Not
```
*bbcbasic
```
As BBC BASIC requires a full 64K segment, won't run in the memory we allocate for MOS start commands.

ZbigniewB, 2023-11-16:
Just updated Agon with newest files for its firmware — and with newest available BBCBASIC — and I can confirm I don't see that flaw anymore (at least at the moment).

breakintoprogram, 2023-11-16:
> Just updated Agon with newest files for its firmware — and with newest available BBCBASIC — and I can confirm I don't see that flaw anymore (at least at the moment).

Cool. I'm pretty sure that it's good, as I use that all the time whilst developing. I'll close this card for now. If it reoccurs, then raise a new one.

closed 2023-11-16 by breakintoprogram